stickbreaker/arduino-esp32 #11: Core dump when doing I2C Address scan

lonerzzz, 2017-11-29:
### Hardware:
Board:							Custom
Core Installation/update date:		2017/11/29
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I managed to find another issue today. When I do a scan of the I2C bus for addresses, I get a core dump and it points to this line (around 852) of the i2c hal file. I checked and byteCnt =1 and queuePos = 0 in this situation.

    if(p_i2c->byteCnt > p_i2c->dq[p_i2c->queuePos].queueLength){// simulate Trans_start

Does the code rely on the receipt of a byte that it won't get when scanning?

stickbreaker, 2017-11-29:
@lonerzzz no, I added an example I2C scan sketch in the Wire Library.  It works fine for me.  It is basically
```c++
Wire.beginTransmission(ID);
if(Wire.endTransmission()==0){
  Serial.printf(" 0x%02x present\n",ID);
}
```
the byteCnt=1 shows that it send the ID byte out.
the full dump should show:
```
[E][esp32-hal-i2c.c:560] dumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:561] dumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:562] dumpI2c(): byteCnt=1
[E][esp32-hal-i2c.c:567] dumpI2c(): [0] (ID<<1) W STOP buf@=0xnnnnnnnn, len=0, pos=0, eventH=0x0 bits=0
```
What do these fields say?
```
[E][esp32-hal-i2c.c:555] dumpI2c(): stage=??
[E][esp32-hal-i2c.c:556] dumpI2c(): error=??
[E][esp32-hal-i2c.c:557] dumpI2c(): event=0x3ffe3590 bits=??
```

Chuck.

lonerzzz, 2017-11-29:
False alarm. My bad. I looked deeper since you do have it covered with an example and finally saw that that it was due to a multithreading issue since the failure was happening after some number of individual address checks. The task doing the scan was different than the regular i2c processing task and my scan code was missing the lock request to access the i2c. I had previously put in locks everywhere so was not expecting any threading issues.

closed 2017-11-29 by lonerzzz