stickbreaker/arduino-esp32 #12: different behavior of DS3231 on rev0 and rev1 chips.

everslick, 2017-12-01:
I just made an interesting observation: My DS3231 (RTC) module works on ESP32 rev0, but does NOT on rev1 chips. ￼This observation holds even when I switch from your I2C fork to the origin/master repository.
An at24c32 EEPROM on the very same I2C bus works on both chip revisions.

FYI: I do not use any Wire.endTransmission(false); calls.

stickbreaker, 2017-12-01:
What library are you using?
I don't have a DS3231, Just a DS1307.
Do you get any core dumps.  Or just no response.
My library will dump on Timeout, Arbitration, bus busy.
NAK, is returned as normal.

In my repo, I added a couple of examples to the `Wire.h` library, Scan and EepromSize.  Scan will show all I2C devices, eeprom size will probe 0x50..0x57 for two byte address EEPROMS.  I might have uploaded eepromsize that will show WritePage size also.

Try them see what they do.

If you get a core dump, save it,  It will contain the Silicon date for the I2C statemachine.
SO, If I need differences for V.0 it will be possible to ident.

Chuck.

everslick, 2017-12-01:
FYI: Turns out, that it is a HW problem with the power supply, and not really related to I2C. The rev0 board just happened to have a more stable PSU. Unfortunately the error case was VERY subtle. Normal operation was fine, but the I2C communication with the DS3231 brought it down, while the EEPROM chip on the same module was working flawlessly all the time, no matter what. Despite plenty of headroom from the PSU (rated for 400mA), I think I need bigger Caps on the ESP32 power pads and maybe for the RTC module. I'm closing this issue.

closed 2017-12-01 by everslick

stickbreaker, 2017-12-01:
@everslick Glad you found it. 

Chuck.