stickbreaker/arduino-esp32 #15: i2cDumpDqData() & i2cDumpInts() Enable Core Debug Level "Error"

chinswain, 2017-12-10:
### Hardware:
Board:							lolin32
Core Installation/update date:		10/12/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:

I've purchased an esp32 (loin32) module to replace an Arduino Mega but I'm having problems with i2c (a TI ADS1115, works fine on the Mega). I've just downloaded your library as I noticed the extra debugging you've added in the hopes of finding whats wrong.

After a few seconds of readings it stops working and produces the dump messages below. Any suggestions on how to debug this?

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c

AIN0: 5240
 
AIN0: 4510
 
AIN0: 5302
 
AIN0: 5262

[E][esp32-hal-i2c.c:603] i2cDumpDqData(): Enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:947] i2cDumpInts(): enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:603] i2cDumpDqData(): Enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:947] i2cDumpInts(): enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:603] i2cDumpDqData(): Enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:947] i2cDumpInts(): enable Core Debug Level "Error"
...........
Library used:
https://github.com/adafruit/Adafruit_ADS1X15
i2c Device:
https://www.adafruit.com/product/1085

### Sketch:
```cpp

#include <Wire.h>
#include <Adafruit_ADS1015.h>

Adafruit_ADS1115 ads;

void setup(void)
{
  Serial.begin(115200);
  ads.setGain(GAIN_TWO);
  ads.begin();
}

void loop(void)
{
  int16_t adc0;
  adc0 = ads.readADC_SingleEnded(0);
  Serial.print("AIN0: "); Serial.println(adc0);
  Serial.println(" ");
  delay(1000);
}
```

### Debug Messages:
```
[E][esp32-hal-i2c.c:603] i2cDumpDqData(): Enable Core Debug Level "Error"
[E][esp32-hal-i2c.c:947] i2cDumpInts(): enable Core Debug Level "Error"

```

stickbreaker, 2017-12-10:
@chinswain to see what the actual error was, you will have to enable Core Debug.

There are two ways to do this:

* Change your board selection to "ESP32 Dev Module", then under the Tools Menu of Arduino you will see `Core Debug Level`.  Select `Error`
* Or you can add the menu option to your lolin32 board by added the following lines to: `hardware\espressif\esp32\boards.txt`  About line 290. Just before the row of '###' inside the lolin32 section.
```
lolin32.menu.DebugLevel.none=None
lolin32.menu.DebugLevel.none.build.code_debug=0
lolin32.menu.DebugLevel.error=Error
lolin32.menu.DebugLevel.error.build.code_debug=1
lolin32.menu.DebugLevel.warn=Warn
lolin32.menu.DebugLevel.warn.build.code_debug=2
lolin32.menu.DebugLevel.info=Info
lolin32.menu.DebugLevel.info.build.code_debug=3
lolin32.menu.DebugLevel.debug=Debug
lolin32.menu.DebugLevel.debug.build.code_debug=4
lolin32.menu.DebugLevel.verbose=Verbose
lolin32.menu.DebugLevel.verbose.build.code_debug=5
```

After you have Core Level Debug Enabled, compile it again and post the dump,  Then I can answer your question.

Chuck.

chinswain, 2017-12-11:
Hi Chuck,

I left the ESP32 running with debug enabled, worked fine for about half an hour and then died, a hardware reset is required to restore normal operation.

Here's the error - anything useful?

```[E][esp32-hal-i2c.c:1131] i2cProcQueue():  Gross Timeout Dead st=0x2f0284, ed=0x2f02b6, =50, max=50 error=1
[E][esp32-hal-i2c.c:609] i2cDumpI2c(): i2c=0x3ffc1024
[E][esp32-hal-i2c.c:610] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:611] i2cDumpI2c(): lock=0x3ffc862c
[E][esp32-hal-i2c.c:612] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:613] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:614] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:615] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:616] i2cDumpI2c(): event=0x3ffc5590 bits=0
[E][esp32-hal-i2c.c:617] i2cDumpI2c(): intr_handle=0x3ffc55c0
[E][esp32-hal-i2c.c:618] i2cDumpI2c(): dq=0x3ffc556c
[E][esp32-hal-i2c.c:619] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:620] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:621] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:582] i2cDumpDqData(): [0] 93 R STOP buf@=0x3ffc229c, len=2, pos=0, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:598] i2cDumpDqData(): 0x0000: U.                               55 83 
[E][esp32-hal-i2c.c:941] i2cDumpInts(): row  count   INTR    TX     RX
[E][esp32-hal-i2c.c:944] i2cDumpInts(): [01] 0x0001 0x0002 0x0001 0x0000 0x002f0284
[E][esp32-hal-i2c.c:944] i2cDumpInts(): [02] 0x0003 0x0100 0x0000 0x0000 0x002f02ab

[E][esp32-hal-i2c.c:1131] i2cProcQueue():  Gross Timeout Dead st=0x2f0318, ed=0x2f034a, =50, max=50 error=1
[E][esp32-hal-i2c.c:609] i2cDumpI2c(): i2c=0x3ffc1024
[E][esp32-hal-i2c.c:610] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:611] i2cDumpI2c(): lock=0x3ffc862c
[E][esp32-hal-i2c.c:612] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:613] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:614] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:615] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:616] i2cDumpI2c(): event=0x3ffc5590 bits=0
[E][esp32-hal-i2c.c:617] i2cDumpI2c(): intr_handle=0x3ffc55c0
[E][esp32-hal-i2c.c:618] i2cDumpI2c(): dq=0x3ffc556c
[E][esp32-hal-i2c.c:619] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:620] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:621] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:582] i2cDumpDqData(): [0] 92 W STOP buf@=0x3ffc2322, len=1, pos=1, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:598] i2cDumpDqData(): 0x0000: .                                01 
[E][esp32-hal-i2c.c:941] i2cDumpInts(): row  count   INTR    TX     RX
[E][esp32-hal-i2c.c:944] i2cDumpInts(): [01] 0x0001 0x0002 0x0002 0x0000 0x002f0318
[E][esp32-hal-i2c.c:944] i2cDumpInts(): [02] 0x0004 0x0100 0x0000 0x0000 0x002f0348```

stickbreaker, 2017-12-11:
@chinswain These two dumps are just showing that the I2C bus was busy, The default 50ms timeout expired before the bus recovered. You might try increasing the timeout with `Wire.setTimeOut(2000);` this would set a 2 second timeout.

What device is is at 0x49?

Chuck.

chinswain, 2017-12-11:
@stickbreaker ok, I've increased the timeout to 2000 - Test started, I'll post back results if it gives up later.

Strange, there was a constant stream of errors but they all appeared to be duplicates of the first two dumps. If it's as simple and issue as a timeout, should it be locking up so drastically and not recovering?

stickbreaker, 2017-12-11:
There is a gap between these two calls.  The first timeout ended at tick 0x002f02ab The next one started a 0x002f0318.  109ms apart.  They may not be directly related.  There could be an I2C call between them that starts the tmeout event.  

I just reread the ADS1115 datasheet.  It states that the ADS1115 cannot drive SCL It cannot stretch the clock to cause a timeout.  The only other possibility is noise on the I2C bus.  What size of pullup resistors are you using? 

Do you have anything else connected to the i2C BUS?

Chuck.

chinswain, 2017-12-11:
I'm using an adafruit board which has 10k pullups on both lines. It's breadboarded with 10cm leads between everything. Power to the ADS1115 is via a fully charged 12V lead acid battery through an MS1117 3.3v Regulator. (esp32 is powered via USB @ 1A).

The ADS1115 is on 0x49, no other devices connected. (although for production I want 40 devices!)

Is there anything unexpected in the library? It looks ok to me:
https://github.com/adafruit/Adafruit_ADS1X15/blob/master/Adafruit_ADS1015.cpp

I'm currently reading the ADS1115 100 times a second to try and trip it up but with the timeout set it seems to be stable so far (92500 readings so far).

stickbreaker, 2017-12-11:
10k pullups are kind of weak, are you using the default 100k I2C clock?  With 10k pullups and 3.3v bus voltage, you only have 0.33ma of current to pull the 'highs' The I2C specification from [UM10204](https://www.nxp.com/docs/en/user-guide/UM10204.pdf) page 47 lists the minimum drive current Iol as 3mA.  The ADS1115 spec's it's drive current at this 3mA level.  Personally I would change your pullups to either 3.3k (1mA of drive) for 100kHz or 2.4k (1.375mA) for 400kHz.

Are you also using WiFi? breadboard layouts can sometimes inject RF energy into the bus.  My testcase uses 6 - 24Lcxx eeproms, 2- MCP23008 I/O Expanders 1 DS307 RTC, These are on multiple PCB's connected by 0.050" ribbon cable (18" long with a 12" free tail). 5 of the 24Lcxx are on a 3.3v segment the MCP's and DS are on a 5V segment behind a BS138 mosfet Level converter.  The pullups are 4.7k on the 5V end with 10k pullups to 3.3v on that segment.  So the effective pullups is about 1.39mA.
I have successfully ran it a 800kHz down to 1kHz.

Chuck.

chinswain, 2017-12-12:
I've left everything as standard, stability is the priority for this project so I've left it at 100k.
I'll try and remove them and close the gap so I can use some 3.3k resistors.

I am using WiFi but it's currently off for testing. I'll move everything to something more permanent once I've proven to myself the esp32 can replace the mega.

stickbreaker, 2017-12-12:
@chinswain well, if you have any more problems Just open another issue.

closed 2017-12-12 by stickbreaker

chinswain, 2017-12-12:
I've added an additional 3.3k resistor to each line so it's at 2.4k. I'll leave it running overnight and update tomorrow.

stickbreaker, 2017-12-12:
Great!

chinswain, 2017-12-13:
Hi Chuck - confirmed - a combination of your library and properly specced pull-ups seems to be the winning combination, if I change either it freaks out and cannot recover without a hard reset (I hope this can be fixed, bit of a nightmare for unattended projects!).

Slightly related, can I ask why you suggested 1mA of drive rather than 3ma?

stickbreaker, 2017-12-13:
@chinswain The 3mA limit is the minimum drive current that must be met for the device to be 'I2C' compatible. I don't like to expect any device to run at 100% of rated performance continuously.  You had stated your circuit was using 10k pullups, (0.33mA)  so changing to 3.3k (1mA). this 3x increase in current cased a power consumption increase in the drive sections of your I2C devices in the amount of (I^2 ) * R. where I is current and R is resistance, so Since R is fixed for each device the power changed from (0.33^2 )  * R = 0.109 * R to (1^2) * R or 1.0 * R increasing to 3mA become 9 * R.  The current increase from 0.33ma to 3.0mA is an ~10fold increase but 0.109R -> 9R is an 82 fold power increase.  The benefit of the extra 2mA was not worth the load.  More just makes the drivers work harder, and the increase in current does not translate into an increase in performance.  The parasitic capacitance and the clock speed are the design considerations (controlling factors) for choosing the pullup values.  
This webpage: [RC Discharging Circuit Tutrorial](http://www.electronics-tutorials.ws/rc/rc_2.html)
goes into the mathematics. I2C signals are triggered at the 30%, 70% voltage points along the curve.  The formula is an exponential function Vc =Vs * e^(-t/(r * c)) Where Vc is the voltage on the buss, Vs is supply (3.3v), t is time(seconds), r is resistance(ohms), c is capacitance(Farad). You can almost solve this equation, But, that requires to you measure 'c' which is really tough on a digital circuit. 

If you start selecting values:
the rise time for the voltage to get to a logic High(2.3v) 70% of Vdd (3.3v).  with a SCL (clock rate) of 100,000Hz to 1/10 of a clock period, t becomes 1/1,000,000 or 0.0000001S
Vc = 2.3v
Vs = 3.3v
e = 2.71828
r=3,300 ohm
Solve the equation for 'c'
c=-t/(r * ln(Vc/Vs))
c=8.39 * 10^-10 F or 839pF 
So as long as my bus capacitance is less than ~1nf, 1mA of drive current will meet my design goals (10% rise time on a 100kHz clock).

I use 0.050" ribbon cable for my I2C interconnects. Most 28ga flat ribbon cable spec out 15pF/ft. I generally increase it an order of magnitude to 150pf/ft this takes into account the connectors, bends, surrounding objects.  So with these rough guesses a 1mA drive current will handle 1m (3ft) of ribbon cable and still meet my 100kHz timing constraints.  If I go to 400kHz or 800kHz. I have to adjust things.

Chuck.

chinswain, 2017-12-13:
@stickbreaker Wow, what an answer! 🥇 
I have quite a bit of reading to do to get these 40 devices working...

Are your changes going to be pulled into the main repo?

stickbreaker, 2017-12-13:
@me-no-dev Stated about two weeks ago he was reviewing it for inclusion, But My coding style does not match the aesthetics of the project 😀 .  So, he was adjusting the spacing and layout.  But, he is in the middle of moving his office to a new Country. ( I think one of the Slavic nations?) so it could be delayed. 

Before this Fork is merged, the debug and testing routines need to be excised.  I am a little verbose in my messages,  😏 .  

If you are going to attach 40 devices to the bus, you will need to actually review(measure) the signals.  You can use a 'scope to calculate 'c' from the curve. 
[image]
This pic shows a START on my test rig.  You can see the exponential decay on the rising edge,  Those overshoots and ringing are cause by my 12" unterminated flying end of the cable. you can see the cross talk at every negative edge. Based on the 5tau infinity assumption, the 5tau is about 1.3us. currently I have 2.48k of resistance on the bus so 1.3us = 5(2480 * c), therefore c = 104pf 
[image]

Chuck.