stickbreaker/arduino-esp32 #16: How to include this in Platformio ?

glucee, 2017-12-12:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:	       11/Dec/2017
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					9600

### Description:
How to easily include this one in the platformio ?

stickbreaker, 2017-12-12:
I have no experience with platformio. This is a fork of arduino-esp32. So, whatever procedures are used to use espressif/arduino-esp32 will work with this fork.  

If you use the `Wire()` library there should not be any differences.

Chuck.

beegee-tokyo, 2017-12-14:
@stickbreaker 
the usual steps would be to put the platform path into platformio.ini of the project.
**`platform = https://github.com/stickbreaker/arduino-esp32.git`**
But unfortunately that fails with some weird error messages. Maybe because your repo is a fork. Don't really know why.
@glucee 
**_2 possible solutions:_**
**a) just replace** 
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c`**
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.h`**
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-log.h`**
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src\Wire.cpp`**
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src\Wire.h`**
with the same files from stickbreakers repo.
**b) use git to clone stickbreakers repo**
rename 
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32`**
to 
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32.org`**
clone stickbreakers repo into an empty 
**`C:\Users\beegee\.platformio\packages\framework-arduinoespressif32`**
then clean your project, recompile and you are using this repo.

stickbreaker, 2017-12-14:
@beegee-tokyo Thanks for the help! 👍 

Chuck.

eohan, 2018-02-15:
Some files that are in the standard repo are not present in stickbreakers' repo. So another solution is to copy stickbreakers files and overwrite all the files in C:\Users\beegee\.platformio\packages\framework-arduinoespressif32 folder. This keeps all the extra files as is.

beegee-tokyo, 2018-02-15:
Not sure if this is still valid, but according to this https://github.com/espressif/arduino-esp32/issues/811 all you have to do is to grab the following files from stickbreakers repo 
```
\libraries\Wire\scr\Wire.h
\libraries\Wire\scr\Wire.cpp
\cores\esp32\esp32-hal-i2c.h
\cores\esp32\esp32-hal-i2c.c
\cores\esp32\esp32-hal-log.h
```
and overwrite the according files in your **`framework-arduinoespressif32`** folder.

closed 2018-10-06 by stickbreaker