stickbreaker/arduino-esp32 #22: I2C Timeout with a PCA9685

holema, 2018-02-14:
### Hardware:
Board:							ESP32 Himalaya
Core Installation/update date:			14.2.2018
IDE name:							Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hello, I use the ESP32 with an I2C PWM Expander PCA9685. With the ESP8266 I had no problem with this IC. But with the ESP32 I get an timeout, but only with this IC. On the same Board there is an LM75 Temp-Sensor, that delivers right values if the PCA is disassembled.
I use this Library:
https://github.com/adafruit/Adafruit-PWM-Servo-Driver-Library/blob/master/Adafruit_PWMServoDriver.cpp library to controll the PCA9685

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
Wire.begin(sda,scl);
ledDriver = Adafruit_PWMServoDriver();
ledDriver.setPWMFreq(200);  
ledDriver.reset();
}

void loop() {
for (uint8_t x = 0; x < 3; x++) {
     ledDriver.setPWM((i+x),0,4000);

    }
    i = 3;
    for (uint8_t x = 0; x < 3; x++) {
      ledDriver.setPWM((i+x),0,4000);

    }
    i = 6;
    for (uint8_t x = 0; x < 3; x++) {
   ledDriver.setPWM((i+x),0,4000);

    }
}
```

### Debug Messages:
```
> Setting PWM 8: 0->4000
[E][esp32-hal-i2c.c:1151] i2cProcQueue():  Busy Timeout start=0xe430, end=0xe462, =50, max=50 error=4
[E][esp32-hal-i2c.c:609] i2cDumpI2c(): i2c=0x3ffc185c
[E][esp32-hal-i2c.c:610] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:612] i2cDumpI2c(): lock=0x3ffd38b4
[E][esp32-hal-i2c.c:614] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:615] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:616] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:617] i2cDumpI2c(): error=4
[E][esp32-hal-i2c.c:618] i2cDumpI2c(): event=0x3ffce908 bits=200
[E][esp32-hal-i2c.c:619] i2cDumpI2c(): intr_handle=0x3ffce5b4
[E][esp32-hal-i2c.c:620] i2cDumpI2c(): dq=0x3ffce590
[E][esp32-hal-i2c.c:621] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:622] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:623] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:582] i2cDumpDqData(): [0] 80 W STOP buf@=0x3ffc37b6, len=5, pos=5, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:598] i2cDumpDqData(): 0x0000: &....                            26 00 00 a0 0f
[E][esp32-hal-i2c.c:944] i2cDumpInts(): row  count   INTR    TX     RX
[E][esp32-hal-i2c.c:947] i2cDumpInts(): [01] 0x0001 0x0002 0x0006 0x0000 0x0000e430
[E][esp32-hal-i2c.c:947] i2cDumpInts(): [02] 0x0004 0x0100 0x0000 0x0000 0x0000e458
```

stickbreaker, 2018-02-19:
@holema This error is hard. See Issue #8 for a discussion.  
The only solution I have found is a hardware reset. A reboot, does not fix it.

Chuck.

stickbreaker, 2018-03-11:
@holema I think I have implemented a fix for this issue.  
Try this branch [stickbreaker-Busy-Glitch](https://github.com/stickbreaker/arduino-esp32/tree/stickbreaker-Busy-Glitch).

You just need to replace:
\cores\esp32\esp32-hal-i2c.c
\cores\esp32\esp32-hal-i2c.h
\cores\esp32\esp32-hal-log.h
\libraries\Wire\src\Wire.h
\libraries\Wire\src\Wire.cpp

Chuck.

closed 2018-03-20 by stickbreaker