stickbreaker/arduino-esp32 #24: Wire.lastError() throwing back error 5

euquiq, 2018-03-10:
Hi Stickbreaker !!!

How are you ? Hope you remember me, I had serious problems with ESP32 and official arduino branch regarding i2c.

Your work saved me! 

And then, today, I decided to update into the latest espressif/arduino-esp32 code from github.

Then I downloaded your branched - i2c enhanced code from git, and copied over the relevant files, into the master esp32 arduino:

\libraries\Wire\scr\Wire.h
\libraries\Wire\scr\Wire.cpp
\cores\esp32\esp32 - hal - i2c.h
\cores\esp32\esp32 - hal - i2c.c
\cores\esp32\esp32 - hal - log.h

I recompiled, and to my dismay, my code is not being able to read 256 bytes from my sensor's EEPROM (which in the older commit was OK until now (I was using ge5b2c1cf).

This is my code (which was working fine):

'''
//Levanto toda la info de la EEPROM del sensor para manejarla en ram (más rápido)
void readEEPROM() { // Read in blocks of 32 bytes to accomodate Wire library
	int err = 0;
	do {
		Wire.beginTransmission(MLX90621_EEPROM);
		Wire.write(0);
		err = Wire.transact(eepromData, 256); //grab 256 byte directly into eepromData
		if (err != 256) { //Error: NO recibí los 256 bytes de la eeprom
			Serial.println("EEPROM ERROR " + (String)Wire.lastError() + ", " + (String)err + " bytes");
			Wire.endTransmission(true); //Cierro la transmision correctamente.
			Wire.reset(); //Y reseteo el interface
		}
	} while (err != 256);
	reseteo = 0; //No da error el i2c
}
'''

Which now keeps in loop throwing at me: 

EEPROM ERROR 5, 0 bytes

I was hoping you may have an idea on what may be going on ? I could try to revert into earlier commits, until I get a working one again, I suppose. But maybe it is an easy thing to fix from my side ?

Regards, 

Enrique.

euquiq, 2018-03-10:
I set up Debug level to ERROR and got this, in case you can use it:

```
[E][esp32-hal-i2c.c:1153] i2cProcQueue():  Busy Timeout start=0x2cd, end=0x318, =75, max=75 error=0
[E][esp32-hal-i2c.c:611] i2cDumpI2c(): i2c=0x3ffc10ec
[E][esp32-hal-i2c.c:612] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[E][esp32-hal-i2c.c:614] i2cDumpI2c(): lock=0x3ffcda60
[E][esp32-hal-i2c.c:616] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:617] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:618] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:619] i2cDumpI2c(): error=0
[E][esp32-hal-i2c.c:620] i2cDumpI2c(): event=0x3ffcdaf8 bits=0
[E][esp32-hal-i2c.c:621] i2cDumpI2c(): intr_handle=0x3ffcdb28
[E][esp32-hal-i2c.c:622] i2cDumpI2c(): dq=0x3ffcdac0
[E][esp32-hal-i2c.c:623] i2cDumpI2c(): queueCount=2
[E][esp32-hal-i2c.c:624] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:625] i2cDumpI2c(): byteCnt=0
[E][esp32-hal-i2c.c:584] i2cDumpDqData(): [0] a0 W  buf@=0x3ffc4302, len=1, pos=1, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0000: .                                00 
[E][esp32-hal-i2c.c:584] i2cDumpDqData(): [1] a1 R STOP buf@=0x3ffc36c4, len=256, pos=0, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0000: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0020: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0040: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0060: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x0080: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x00a0: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x00c0: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:600] i2cDumpDqData(): 0x00e0: ................................ 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E][esp32-hal-i2c.c:946] i2cDumpInts(): row  count   INTR    TX     RX
[E][esp32-hal-i2c.c:949] i2cDumpInts(): [01] 0x0001 0x0002 0x0003 0x0000 0x000002cd
[E][esp32-hal-i2c.c:949] i2cDumpInts(): [02] 0x0001 0x0200 0x0000 0x0000 0x000002cd
````
Enrique

stickbreaker, 2018-03-11:
@euquiq 
It could be the changes I made to remove the signal glitch when Wire() attaches the i2c peripheral to the bus.

I made another attempt.  Try this branch: [stickbreaker-Busy-Glitch](https://github.com/stickbreaker/arduino-esp32/tree/stickbreaker-Busy-Glitch).

It has recovered bus busy failures for me.

Chuck.

stickbreaker, 2018-03-11:
@euquiq 
Don't use this new branch, (stickbreaker-Busy-Glitch).  I just found some more problems.  I just proved that my hardware reset works, but, I forgot to restore all of the configuration values.  They all went back to Reset State(hooray!) 

Chuck.

euquiq, 2018-03-11:
Hi Chuck! No Sweat, Thanks a lot, I hope you can nail this glitch, whenever you got time. If I can be of any help, just ask !

stickbreaker, 2018-03-11:
@euquiq 
   It should be up.  Try this branch [stickbreaker-Busy-Glitch](https://github.com/stickbreaker/arduino-esp32/tree/stickbreaker-Busy-Glitch)

Chuck.

stickbreaker, 2018-03-11:
@lonerzzz @me-no-dev Just figured out how to do a hardware reset on the i2c stateMachine.  
I have been able to recover from grounding SCL and SDA!

Chuck.

euquiq, 2018-03-11:
Hi Chuck,

Again, thanks for your time and help.

Just downloaded branch stickbreaker-Busy-Glitch and copied into my esp32 branch your usual files:

\libraries\Wire\scr\Wire.h
\libraries\Wire\scr\Wire.cpp
\cores\esp32\esp32 - hal - i2c.h
\cores\esp32\esp32 - hal - i2c.c
\cores\esp32\esp32 - hal - log.h

I got the following errors when compiling:

Compiling 'ESP32_MLX90621_GAS_MULTITASK_WEB' for 'DOIT ESP32 DEVKIT V1'
Wire.cpp: In member function void TwoWire::begin(int, int, uint32_t)
Error compiling libraries
End
Build failed for project 'ESP32_MLX90621_GAS_MULTITASK_WEB'
 
Wire.cpp: 71:26: error: 'i2cDetatchSCL' was not declared in this scope
   i2cDetatchSCL(i2c,scl); \\ detach pins before resetting I2C perpherial
 
Wire.cpp: 72:26: error: 'i2cDetatchSDA' was not declared in this scope
   i2cDetatchSDA(i2c,sda); \\ else a glitch will appear on the i2c bus

stickbreaker, 2018-03-11:
@euquiq that doesn't make any sense, I'll look!

stickbreaker, 2018-03-11:
@euquiq 
They are right where they are suppose to be?

[i2cDetachSCL()](https://github.com/stickbreaker/arduino-esp32/blob/6b682d23e581f9669cdec11faf2825c34c76e6ba/cores/esp32/esp32-hal-i2c.c#L185url)
[i2cDetachSDA](https://github.com/stickbreaker/arduino-esp32/blob/6b682d23e581f9669cdec11faf2825c34c76e6ba/cores/esp32/esp32-hal-i2c.c#L208)

```c++

i2c_err_t i2cAttachSCL(i2c_t * i2c, int8_t scl)
{
    if(i2c == NULL){
        return I2C_ERROR_DEV;
    }
    digitalWrite(scl, HIGH); 
    pinMode(scl, OPEN_DRAIN | PULLUP | INPUT | OUTPUT);
    pinMatrixOutAttach(scl, I2C_SCL_IDX(i2c->num), false, false);
    pinMatrixInAttach(scl, I2C_SCL_IDX(i2c->num), false);
    return I2C_ERROR_OK;
}

i2c_err_t i2cDetachSCL(i2c_t * i2c, int8_t scl)
{
    if(i2c == NULL){
        return I2C_ERROR_DEV;
    }
    pinMatrixOutDetach(scl, false, false);
    pinMatrixInDetach(I2C_SCL_IDX(i2c->num), false, false);
    pinMode(scl, INPUT | PULLUP);
    return I2C_ERROR_OK;
}

i2c_err_t i2cAttachSDA(i2c_t * i2c, int8_t sda)
{
    if(i2c == NULL){
        return I2C_ERROR_DEV;
    }
    digitalWrite(sda, HIGH); 
    pinMode(sda, OPEN_DRAIN | PULLUP | INPUT | OUTPUT );
    pinMatrixOutAttach(sda, I2C_SDA_IDX(i2c->num), false, false);
    pinMatrixInAttach(sda, I2C_SDA_IDX(i2c->num), false);
    return I2C_ERROR_OK;
}

i2c_err_t i2cDetachSDA(i2c_t * i2c, int8_t sda)
{
    if(i2c == NULL){
        return I2C_ERROR_DEV;
    }
    pinMatrixOutDetach(sda, false, false);
    pinMatrixInDetach(I2C_SDA_IDX(i2c->num), false, false);
    pinMode(sda, INPUT | PULLUP);
    return I2C_ERROR_OK;
}
```

Chuck

stickbreaker, 2018-03-11:
@euquiq Manual Spelling Error~!!!!

Change them to i2cDetachSCL() and i2cDetachSDA()  (remove the extra 't')

Chuck.

euquiq, 2018-03-11:
Works fine !!  Thanks again ! You are a great contributor to the Arduino side of the ESP32 !!

BTW, I've been reading your opinions / concerns about ESP32 Arduino framework development (in here https://github.com/stickbreaker/arduino-esp32/issues/21) and I must say I  am a bit appalled about the lethargic situation too.

A couple of weeks ago, the ESP8266 finally got a WebServerSecure library and examples. So the esp8266 is capable of serving thru https.  

About a year ago, I dumped the esp8266 and jumped into the ESP32 because, among other things, I assumed it was the logical step forward for delving into HTTPS server.

But then: https://github.com/esp8266/Arduino/tree/bd1c7ce1dc4f56d707637f2829f6473e888e533d/libraries/ESP8266WebServer

closed 2018-03-11 by euquiq

stickbreaker, 2018-03-11:
@euquiq glad to hear it is now working for you.  I'll merge it into the main repo later today.

I don't know why arduino-esp is being treated as the bastard stepchild. It seems to me it should be the path forward :)

Chuck.