stickbreaker/arduino-esp32 #26: Small compilation warning with I2C patched file

rrobinet, 2018-03-13:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 WEMOS LOLIN32 WROOM
Core Installation/update date:		        11/jan/2017 (last version from Github
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

Hi stickbreaker,

I have a small issue while compiling your I2C patched version, and I could't find the exact reason.
May be this warning can be corrected but I don't know how:
The message is the following:
... /hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c: In function 'i2cDumpDqData':
.../hardware/espressif/esp32/cores/esp32/esp32-hal-i2c.c:603:1: warning: implicit declaration of function 'log_n' [-Wimplicit-function-declaration]
 log_n("Enable Core Debug Level \"Error\"");

Can you explain this.
Thanks in avance
Robert

rrobinet, 2018-03-13:
Please ignore this issue
After reloading your last save set everything was OK
Sorry
Robert

closed 2018-03-13 by rrobinet

stickbreaker, 2018-03-13:
@rrobinet are you running [patch-1](https://github.com/stickbreaker/arduino-esp32/tree/patch-1) or [stickbreaker-Busy-Glitch](https://github.com/stickbreaker/arduino-esp32/tree/stickbreaker-Busy-Glitch)?

Patch-1 is just the espressif main arduino branch with a hardware reset of the i2c peripherals.

stickbreaker-Busy-Glitch is the current version of my interrupt driven library.

Busy-Glitch is what I would recommend.

Chuck.

rrobinet, 2018-03-14:
Chuck,
I had a previous version of your code: 
(https://github.com/stickbreaker/arduino-esp32)
Now because there is another issue with the UART, see https://github.com/espressif/arduino-esp32/issues/1189, I had to patch the uartFlush(uart_t* uart) function of the file  esp32_hal_uart.c 
This one was updated from your arduino-esp32 version I had. So I did try to merge your I2C files with the last version of the arduino-sep32 save set and the UART patch. Something went probably wrong so the reason of the warning message.
Finally I did reinstall your code https://github.com/stickbreaker/arduino-esp32 and then modify the  esp32_hal_uart.c  file, and I don't have problem anymore.
NB:
It becomes quite cumbersome to manage all these unofficial patches, I hope that some action will be taken by the ESP32 Esspresive team to include all these modifications in the official release.
Robert

stickbreaker, 2018-03-14:
I understand. I'm currently working with 3 different versions of arduino-esp32. I have wasted too much time modifying the 'wrong' one!

Chuck