stickbreaker/arduino-esp32 #28: Problem in i2C between ESP32 and Adafruit_ADS1015

Natalie88, 2018-03-28:
### Hardware:
Board: 	ESP32  						?ESP32 
Core Installation/update date:			?27/Mar/2018?
IDE name:							?Arduino IDE
Flash Frequency:					?80Mhz?
Upload Speed:						?115200?

### Description:
I have ESP32 connected to ADS1015 through i2C (PIN 18, 19), everything was cool when I first had the ESP8266  but when I changed to ESP32 first I got zeros from the function readADC_SingleEnded()
However when I changed to your arduino-esp32 I am getting a constant value of 4096 all the time. 
any suggestions. 

#include "Arduino.h"
#include "Adafruit_ADS1015.h"
Adafruit_ADS1015 adc;

void getSolarCurrent() {
  adc.setGain(adsGain_t::GAIN_TWO); //set upper limit vmax to 2.048V
  uint16_t adc_res = adc.readADC_SingleEnded(1);
  Serial.println("ADC res: "+String(adc_res));
  Serial.flush();
  return (double) adc_res / 8.326; //Result is in mA (adc_val 2048mV/2047 / 8.33 / 1 Ohm)

}
void getCapVoltage() {
  adc.setGain(adsGain_t::GAIN_ONE); //set vmax to 4.096V
  uint16_t adc_res = adc.readADC_SingleEnded(0);
  Serial.println("ADC res: "+String(adc_res));
  Serial.flush();
  return (double)adc_res / 0.4998;//Result in mV (adc_res * 4096mV/2047)

}

void setup() {
   //rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);   
  Serial.begin(115200);
  Serial.println();
  Serial.println("Booted");
  Serial.flush();

    adc.begin(); /* own begin-method which sets sda=2 and scl=14 pins to the ones used on solar board (and data sheet) */
    
    yield();  
  
    getSolarCurrent();
    getCapVoltage();
    esp_sleep_enable_timer_wakeup(10*1000000);
    
    esp_deep_sleep_start();   
}

void loop() {

}
### Debug Messages:
```
Booted
ADC res: 4095
ADC res: 4095

Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2018-03-28:
@Natalie88 , You 'scope image does not show a correctly functioning i2c bus.  SCL should be a regular square wave, oscillating from 3.3v to 0v.  If I am reading your image correctly it has an irregular pattern that is only varying from 3v to 3.3v.  SDA is also wrong, It should be idle at 3.3v not 0V.

 I don't see where you specify the SDA and SCL pins using `Wire.begin();`

With my current release (V0.2.0) any call to `Wire.begin();` without specifying the sdaPin and sclPin will default back to SDA and SCL as defined by the board variant. If you have selected just the generic 'esp32' the compiler will use this file:
*..hardware\espressif\esp32\variants\esp32\pins_arduino.h*  which defines SDA as 21 and SCL as 23.

Search through your code for `Wire.begin()`, make sure that ALL of them specify your nonstandard pin assignment. 

I have proposed a path in the main Espressif repo [1239](https://github.com/espressif/arduino-esp32/pull/1239) that fixes this problem.  I haven't yet posted it to my repo.

Chuck.

Natalie88, 2018-03-28:
Thank you very much for your reply i have configured the PINs inside the ADS_1015 library in the begin() method i altered it instead of wire.begin() to wire.bigen(18,19)

stickbreaker, 2018-03-29:
@natalie88 try the scan.ino example. Update the wire.begin() to match your hardware. That will show all devices on the bus. 

The 4095 probably is just the 'no comms' value. Lets verify the esp32 can see the adc. What resistance value are your pullups?

Chuck

Natalie88, 2018-03-30:
@stickbreaker I got it to work finally, the problem was hardware related. 
Thank you very much