stickbreaker/arduino-esp32 #31: Bus busy error when using 2nd TwoWire instance

chinswain, 2018-04-27:
### Hardware:
Board:	LOLIN32
Core Installation/update date:	Main + latest stickbreaker patched i2c files
IDE name:	Arduino IDE
Flash Frequency:	40Mhz
Upload Speed:	115200

### Description:
Just testing the second i2c on the ESP32 with the i2c changes, seems there's an issue having both active?
I've modified the scanner example to scan both, getting a busy bus error. No issues when scanning only one bus in the loop, even when beginning both.

```
[I][esp32-hal-i2c.c:971] i2cProcQueue(): Bus busy, reinit
```

### Sketch:
```cpp

#include <Wire.h>

TwoWire Wire1(1);

void scan() {
  Serial.println("\n Scanning I2C Addresses");
  uint8_t cnt = 0;
  for (uint8_t i = 0; i < 0x7F; i++) {
    Wire.beginTransmission(i);
    uint8_t ec = Wire.endTransmission(true); // if device exists on bus, it will aCK
    if (ec == 0) { // Device ACK'd
      if (i < 16)Serial.print('0');
      Serial.print(i, HEX);
      cnt++;
    }
    else Serial.print(".."); // no one answered
    Serial.print(' ');
    if ((i & 0x0f) == 0x0f)Serial.println();
  }
  Serial.print("Scan Completed, ");
  Serial.print(cnt);
  Serial.println(" I2C Devices found.");
}

void scan2() {
  Serial.println("\n Scanning I2C Addresses");
  uint8_t cnt = 0;
  for (uint8_t i = 0; i < 0x7F; i++) {
    Wire1.beginTransmission(i);
    uint8_t ec = Wire1.endTransmission(true); // if device exists on bus, it will aCK
    if (ec == 0) { // Device ACK'd
      if (i < 16)Serial.print('0');
      Serial.print(i, HEX);
      cnt++;
    }
    else Serial.print(".."); // no one answered
    Serial.print(' ');
    if ((i & 0x0f) == 0x0f)Serial.println();
  }
  Serial.print("Scan Completed, ");
  Serial.print(cnt);
  Serial.println(" I2C Devices found.");
}

void setup() {
  Serial.begin(115200);
  Wire.begin();
  Wire1.begin(19, 23);

}

void loop() {
  scan();
  scan2();
  delay(500);
}
```

### Debug Messages:
```
 Scanning I2C Addresses
00 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. 48 .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. 57 .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. 68 .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. Scan Completed, 4 I2C Devices found.

 Scanning I2C Addresses
[I][esp32-hal-i2c.c:971] i2cProcQueue(): Bus busy, reinit
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
20 .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. .. 
.. .. .. .. .. .. .. .. .. .. .. .. .. .. .. Scan Completed, 1 I2C Devices found.

```

chinswain, 2018-05-12:
Hi Chuck, finally got myself a scope, one of the slaves was malfunctioning - replaced with identical part and no more time outs.

closed 2018-05-12 by chinswain

stickbreaker, 2018-05-12:
Yea!