stickbreaker/arduino-esp32 #32: Compiling on Arduino IDE 1.8.5 with maximum compiler warnings causes compilation to fail

PicoEmma, 2018-05-17:
Compiling on Arduino IDE 1.8.5 with maximum compiler warnings causes compilation to fail. (warnings treated as errors).

Seems to be exclusively an issue in the EEPROM files

console output:

```cpp
C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'int8_t EEPROMClass::readChar(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:188:46: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   return EEPROMClass::readAll (address, value);

                                              ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'uint8_t EEPROMClass::readUChar(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:194:46: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   return EEPROMClass::readAll (address, value);

                                              ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'int16_t EEPROMClass::readShort(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:199:11: note: 'value' was declared here

   int16_t value;

           ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'uint16_t EEPROMClass::readUShort(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:205:12: note: 'value' was declared here

   uint16_t value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'int32_t EEPROMClass::readLong(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:223:11: note: 'value' was declared here

   int32_t value;

           ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'uint32_t EEPROMClass::readULong(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:229:12: note: 'value' was declared here

   uint32_t value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'int64_t EEPROMClass::readLong64(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:235:11: note: 'value' was declared here

   int64_t value;

           ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'uint64_t EEPROMClass::readULong64(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:241:12: note: 'value' was declared here

   uint64_t value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'float_t EEPROMClass::readFloat(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:247:11: note: 'value' was declared here

   float_t value;

           ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'double_t EEPROMClass::readDouble(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:317:12: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

     return value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:253:12: note: 'value' was declared here

   double_t value;

            ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: In member function 'bool EEPROMClass::readBool(int)':

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:260:54: error: 'value' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   return EEPROMClass::readAll (address, value) ? 1 : 0;

                                                      ^

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp: At global scope:

C:\Users\emmab\Documents\Arduino\hardware\espressif\esp32\libraries\EEPROM\EEPROM.cpp:32:20: warning: 'TAG' defined but not used [-Wunused-variable]

 static const char* TAG = "eeprom";

                    ^

cc1plus.exe: some warnings being treated as errors
```

stickbreaker, 2018-05-18:
@PicoEmma you'll want to post this issue on espressif/arduino.  I am only using this fork to improve i2c communications.  There are only five files that I am modifiying:
>../libraries/Wire/Wire.h
../libraries/Wire/Wire.cpp
../cores/esp32/esp32-hal-i2c.h
../cores/esp32/esp32-hal-i2c.c
../cores/esp32/esp32-hal-log.h

All of the errors you posted refer to:
>../libraries/EEPROM/EEPROM.cpp

So, any changes would be lost when my fork is merged back into espressif/arduino.

Chuck.

PicoEmma, 2018-05-18:
After fixing the EEPROM issues the compiler gets into the i2c files as well

example:

```
C:\Users\e\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c:1088:5: error: 'ret' may be used uninitialized in this function [-Werror=maybe-uninitialized]

   if(ret!=ESP_OK){

```

stickbreaker, 2018-05-20:
@PicoEmma Don't apologize for my mistakes.
I fixed most of those errors, but, three still exist. I initialize a variable but only use it in a print statement, the compiler complains even though I am using the value:
```
C:\Users\e\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c: In function 'dumpCmdQueue':

C:\Users\e\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c:399:17: warning: variable 'c' set but not used [-Wunused-but-set-variable]

   I2C_COMMAND_t c;

                 ^

C:\Users\e\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c:1111:14: warning: unused variable 'tAfter' [-Wunused-variable]

 portTickType tAfter=xTaskGetTickCount();

              ^

C:\Users\e\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c:1101:14: warning: unused variable 'tBefore' [-Wunused-variable]

 portTickType tBefore=xTaskGetTickCount();  

              ^
```

Do you know how to work around these warnings?

Chuck.

PicoEmma, 2018-05-21:
For that first one I see that there is a possibility that the print statement does not get defined:

```cpp
#if ARDUHAL_LOG_LEVEL >= ARDUHAL_LOG_LEVEL_ERROR
#define log_e(format, ...) log_printf(ARDUHAL_LOG_FORMAT(E, format), ##__VA_ARGS__)
#else
#define log_e(format, ...)  // no definition
#endif
```

I'm not sure I have the same files and branch as you have so I don't trust what I'm getting in compilation. But I believe that ensuring that all code paths use the variable will fix the "not used" warning? Please correct me if I am wrong.

closed 2018-10-06 by stickbreaker