stickbreaker/arduino-esp32 #34: Compilation error with recent changes

chinswain, 2018-05-20:
I've just refreshed the i2c files from your fork and can no longer compile.

```
D:\OneDrive\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c: In function 'i2cInit':

D:\OneDrive\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-i2c.c:329:5: warning: implicit declaration of function 'i2cReleaseISR' [-Wimplicit-function-declaration]

     i2cReleaseISR(i2c); // ISR exists, release it before disabling hardware

```

stickbreaker, 2018-05-20:
@chinswain , I did a direct file edit and screwed up,  I think I fixed it, but I'm not at my dev pc, so I haven't compiled it.

Sorry, this evening I'll try it.

Just revert to 0.2.2

Chuck.

stickbreaker, 2018-05-24:
@chinswain 

I just figured out where I screwed up,  when I tried to clear compiler warnings, I added the `if()` code.  Bad Me!

Change code to this: esp-hal-i2c.c:1028
```c++
if(i2c->i2c_event) {
  xEventGroupClearBits(i2c->i2c_event, 0xFF);
  }
else {// failed to create EventGroup
```

I should not have tried to clean up the compiler warnings without actually testing the result.  I haven't be back to my dev system for 3 weeks.

Sorry.

Chuck.

chinswain, 2018-05-24:
Made the change - confirmed fixed.

I think you deserved a break away from this.. It's been an adventure I'm sure!

closed 2018-05-24 by chinswain

me-no-dev, 2018-05-24:
@stickbreaker which line corresponds to this change in the official repo branch?

stickbreaker, 2018-05-24:
@me-no-dev , not in stickbreaker-i2c, only in my repo main.

PhilColbert, 2018-05-30:
Hi,

Has this been fixed now, just updated to your latest code and same error

error: implicit declaration of function 'i2cReleaseISR' [-Werror=implicit-function-declaration]
     i2cReleaseISR(i2c); // ISR exists, release it before disabling hardware

thanks again.

On another note I just updated to the latest espidf and arduino as a component and suddenly hundreds of errors started coming out, hence trying to update to your latest...... the newest espidf code crashes my project in all different kinds of ways now :( sad....