stickbreaker/arduino-esp32 #35: Grosstimeout dead?

PhilColbert, 2018-05-30:
Any ideas on this one, appears every minute or so ?

It still works pretty well, just be nice to fix this error :)

Thanks.

[1B][0;31m[E][esp32-hal-i2c.c:1220] i2cProcQueue():  Gross Timeout Dead start=0x1e6b2b, end=0x1e6b8f, =100, max=100 error=1[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:638] i2cDumpI2c(): i2c=0x3ffb13e8[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:649] i2cDumpI2c(): dev=0x60013000 date=0x16042000 level=ERROR[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:651] i2cDumpI2c(): lock=0x3ffce1fc[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:653] i2cDumpI2c(): num=0[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:654] i2cDumpI2c(): mode=1[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:655] i2cDumpI2c(): stage=3[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:656] i2cDumpI2c(): error=1[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:657] i2cDumpI2c(): event=0x3ffd2658 bits=0[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:658] i2cDumpI2c(): intr_handle=0x3ffce260[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:659] i2cDumpI2c(): dq=0x3ffd4ce4[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:660] i2cDumpI2c(): queueCount=2[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:661] i2cDumpI2c(): queuePos=1[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:662] i2cDumpI2c(): byteCnt=1[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:612] i2cDumpDqData(): [0] ee W  buf@=0x3ffb87ce, len=1, pos=1, eventH=0x0 bits=0[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:628] i2cDumpDqData(): 0x0000: .                                00 [1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:612] i2cDumpDqData(): [1] ef R STOP buf@=0x3ffb8748, len=3, pos=0, eventH=0x0 bits=0[1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:628] i2cDumpDqData(): 0x0000: .8.                              09 38 ff [1B][0m
[1B][0;31m[E][esp32-hal-i2c.c:982] i2cDumpInts(): enable Core Debug Level "Error"[1B][0m
V (1993971) intr_alloc: esp_intr_free: Disabling int, killing handler[1B][0m
V (1993977) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args[1B][0m
V (1993984) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting f***********lags 0xE[1B][0m***********

stickbreaker, 2018-05-30:
@PhilColbert Base on this dump, you have a device at 0x77:
```
Wire.beginTransmission(0x77);
Wire.write(0x0);
Wire.endTransmission(false);
Wire.requestFrom(0x77,3,true); // This is where the timeout occurs.
```
The device accepts the `Write()`, acknowledges the `ReSTART` `READ()` address, then holds the SCL low.  It is pausing before it sends any data out.  This SCL stretch is longer than 100ms, your current timeout value..

What device is at 0x77? It looks like this 'error' is normal operation of the device.  My i2c is handling it correctly. 

Review the datasheet on the device, verify its timing performance. Maybe you need to allow additional time for its operation.  You can increase the timeout with `Wire.setTimeOut(uint16_t milliseconds);`. Currently you have it set to 100, try 250 or 500.  But, then `Wire.requestFrom()` will hang up to that timeout value if the device holds SCL low.

Chuck.

PhilColbert, 2018-05-30:
Thanks for the info, Its an ms5611 inside a GY-86 device.

Response time is supposed to be 0.5ms at the specified oversampling rate.

Will change the timeout and see what happens, just noticed the serial data clk might be too high, will have a look at that aswell, see if that sorts it.

Thanks again.

By the way, the current code doesnt compile, it needs the function declaration for i2cReleaseISR  to be moved to the top of the file, or above where you added the call to the function for 0.2.3 thanks

chinswain, 2018-05-30:
@PhilColbert See issue #34 

https://github.com/stickbreaker/arduino-esp32/issues/34

PhilColbert, 2018-05-30:
Thanks, I commented at the end of the thread, just changed the timeout to 250 and the freq to 20mhz instead of 40mhz and .... touch wood, no errors yet ! :)

Used to get one every 30 seconds so its looking good.

thanks for your efforts again !!

stickbreaker, 2018-05-30:
@PhilColbert 
>Thanks for the info, Its an ms5611 inside a GY-86 device.
Response time is supposed to be 0.5ms at the specified oversampling rate.
Will change the timeout and see what happens, just noticed the serial data clk might be too high, will have a look at that aswell, see if that sorts it.

Are you sure 0.5ms? The default is 50ms and you had it at 100ms.  Something doesn't seem right.  if you are getting a failure at 100ms and the spec sheet says 0.5ms, that is 200 times longer?  The error handling for gross timeout might need to be changed, instead of creating the debug output just pass the error back to your application and have your application handle it, by either resetting the i2c bus or going into a timeout loop in the app.

What do they say about this sampling cycle failure? Is it a common occurrence with a know solution, or is this a singleton event?  What is your electrical circuit? Pullups?

>By the way, the current code doesnt compile, it needs the function declaration for i2cReleaseISR to be moved to the top of the file, or above where you added the call to the function for 0.2.3 thanks

More sloppy code, my fault.  I'll change that when I get back to my dev system.  I have been bitten too many times to try an ad hock edit directly on GitHub.  

Chuck.

PhilColbert, 2018-05-30:
Well I have changed the frequency to 20mhz from 40 and its not erroring anymore, perhaps thats solved the issue ?

Seems perfect now and much more stable - no errors and the device seems to be spot on.

I'll look into the rest when I get time - trying to get BLE and Bluetooth to be stable on the esp32 at the moment, its needed by my application asap.

Thanks

PhilColbert, 2018-05-31:
Well the new code seems to be much better, no error messages coming out at all, perfect !

Thanks again - just need to have it as standard in the arduino core :)

closed 2018-05-31 by PhilColbert