stickbreaker/arduino-esp32 #37: i2c error causing panic

ayhoung, 2018-06-02:
### Hardware:
Board:							Sparkfun ESP32 Thing
Core Installation/update date:			2/june/2018
IDE name:							Platform.io
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hey all! With the main arduino esp32 project right now, I'm getting the occasional insidious i2c timeout error, which led me to @stickbreaker's project. 

I tried to replace the following files:
```
\libraries\Wire\scr\Wire.h
\libraries\Wire\scr\Wire.cpp
\cores\esp32\esp32-hal-i2c.h
\cores\esp32\esp32-hal-i2c.c
\cores\esp32\esp32-hal-log.h
```

from @stickbreaker's fork into the current `https://github.com/platformio/platform-espressif32.git#feature/stage` build of platform.io

but this resulted in the below error:
```
[E][esp32-hal-i2c.c:1212] i2cProcQueue():  Gross Timeout Dead start=0x120e, end=0x120e, =0, max=51 error=1
[E][esp32-hal-i2c.c:630] i2cDumpI2c(): i2c=0x3ffc1554
[E][esp32-hal-i2c.c:641] i2cDumpI2c(): dev=0x60013000 date=0x16042000 level=ERROR
[E][esp32-hal-i2c.c:643] i2cDumpI2c(): lock=0x3ffcc0a4
[E][esp32-hal-i2c.c:645] i2cDumpI2c(): num=0
[E][esp32-hal-i2c.c:646] i2cDumpI2c(): mode=1
[E][esp32-hal-i2c.c:647] i2cDumpI2c(): stage=3
[E][esp32-hal-i2c.c:648] i2cDumpI2c(): error=1
[E][esp32-hal-i2c.c:649] i2cDumpI2c(): event=0x3ffcc128 bits=10
[E][esp32-hal-i2c.c:650] i2cDumpI2c(): intr_handle=0x3ffcd4a4
[E][esp32-hal-i2c.c:651] i2cDumpI2c(): dq=0x3ffcd294
[E][esp32-hal-i2c.c:652] i2cDumpI2c(): queueCount=1
[E][esp32-hal-i2c.c:653] i2cDumpI2c(): queuePos=0
[E][esp32-hal-i2c.c:654] i2cDumpI2c(): byteCnt=1
[E][esp32-hal-i2c.c:604] i2cDumpDqData(): [0] 78 W STOP buf@=0x3ffc51a2, len=17, pos=17, eventH=0x0 bits=0
[E][esp32-hal-i2c.c:620] i2cDumpDqData(): 0x0000: @..ffaa``........                40 1f 1f 66 66 61 61 60 60 1f 1f 00 00 00 00 00 00
[E][esp32-hal-i2c.c:974] i2cDumpInts(): enable Core Debug Level "Error"
GGuru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x4008431b  PS      : 0x00060031  A0      : 0x800843ba  A1      : 0x3ffc0ba0
A2      : 0x3ffc1554  A3      : 0x00000001  A4      : 0x00000010  A5      : 0x80000020
A6      : 0x00060021  A7      : 0x00000000  A8      : 0xbaad9677  A9      : 0x3ffe2bd0
A10     : 0x000000b7  A11     : 0xbaad5678  A12     : 0x00004000  A13     : 0x3ffd2470
A14     : 0x00000003  A15     : 0x00060c23  SAR     : 0x00000014  EXCCAUSE: 0x0000001d
EXCVADDR: 0xbaad9677  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x4008431b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400dd97a

Backtrace: 0x4008431b:0x3ffc0ba0 0x400843b7:0x3ffc0bc0 0x40084a36:0x3ffc0bf0 0x40081899:0x3ffc0c10 0x400dd977:0x00000000
```
### Sketch:
```cpp
#include <Wire.h>
#include <Adafruit_INA219.h> // You will need to download this library

Adafruit_INA219 sensor219; // Declare and instance of INA219

void setup(void) 
{
      
  Serial.begin(9600);    
  sensor219.begin();
  
}

void loop(void) 
{
  float busVoltage = 0;
  float current = 0; // Measure in milli amps
  float power = 0;

  busVoltage = sensor219.getBusVoltage_V();
  current = sensor219.getCurrent_mA();
  power = busVoltage * (current/1000); // Calculate the Power
  
  
  Serial.print("Bus Voltage:   "); 
  Serial.print(busVoltage); 
  Serial.println(" V");  
  
  Serial.print("Current:       "); 
  Serial.print(current); 
  Serial.println(" mA");
  
  Serial.print("Power:         "); 
  Serial.print(power); 
  Serial.println(" W");  
  
  Serial.println("");  

  delay(2000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2018-06-02:
@ayhoung your sketch doesn't match the error report.

The i2c Gross timeout shows an i2c device at 0x3C is holding SCL low. INA219 possible address range of 0x40 .. 0x4f

Something doesn't match.

Verify you have pullups on SCL and SDA, at least 2.4k ohm.

Chuck

ayhoung, 2018-06-02:
@stickbreaker really appreciate the reply! I'm really sorry it seems the error was originating when I was setting up two i2c devices (SSD1306 / INA219) and I was calling `ina219.begin();` outside of the `setup(){}` clause. I didn't originally suspect this as an issue since it worked fine in the main branch, but moving around the code somewhat I can now get it working with your files :)

closed 2018-06-02 by ayhoung