stickbreaker/arduino-esp32 #39: Understanding merge with Main

jjassar, 2018-07-16:
Hi stickbreaker.
First i want to say thanks for your contribution on I2c issues.
I am beginner so not able to understand one thing properly. Your changes are merged in main repo. but i see two methods named "TwoWire::transact()" in your Wire.cpp but they are not present in latest Repo of ESP32 arduino. So does it means that i still need to replace all 5 files from your release to esp32 arduino core?

stickbreaker, 2018-07-16:
@jjassar when my i2c driver code was merged, the additional functionality i added to `Wire()` were not accepted, the value of these new function was weighted against the confusion factor of adding new methods. It was deemed too confusing for new Arduino programmers.  So they were left out.  I was directed to create a new library to encapsulate these new features.  I haven't had time yet, I have a couple of changes to the i2c driver before I will have time.

Chuck.

jjassar, 2018-07-17:
So it means that, if i need to follow your code then which files i should replace in main repo or should i add "TwoWire::transact()" methods manually in Wire.cpp of main repo?
Is there any other methods which are added and required for reliable i2c operation ?
Please guide the best way for reliability of i2c interface.

stickbreaker, 2018-07-17:
@jjassar you will have to replace the same 5 files, stability should be the same.  I'm working on performance right now. When I complete it, I'll apply it to my repo.

jjassar, 2018-07-19:
@stickbreaker Thanks a lot for your efforts.

closed 2018-07-19 by jjassar