stickbreaker/arduino-esp32 #41: I2C with Adafruit Si7021 humidity/temp sensor

IraSch, 2018-08-13:
### Hardware:
Board:							Oddwires IoT ESP32 Dev Module
Core Installation/update date:			3/3/2018 (Not sure how to get the actual Core date
IDE name:							Arduino IDE v1.8.5
Flash Frequency:					80MHz
Upload Speed:						921600

### Description:
I have been having problems with using I2C on the OddWires IoT ESP32 board for an Adafruit Si7021 humidity/temperature board. It will work for a while (30 minutes to several hours, random), then fail by returning an out of range temperature. From then on, it will report the same temperature until reset. 

I tried first installing (overwriting) esp32-hal-i2c.h, .cpp and esp32-hal-log.h in cores, and wire.h and .cpp in the Wire library.  This compiled but returned a negative temperature value. 

I also tried installing (overwrite) the entire package - same result. 

Am I missing something simple? Seems to be communicating but getting some weird value.

Thanks, Ira
### Sketch:

### Debug Messages:

IraSch, 2018-08-13:
Thanks for the suggestion. I will. Do you happen to know if I need to use the stickbreaker core and wire library changes, or does this Si7021 library work with the original core and library files?

 

Ira

 

Try this modified library https://github.com/stickbreaker/Adafruit_Si7021

stickbreaker, 2018-08-14:
Rc3 + 
But rc4 and release 1.0.0 are broken, waiting on merge of pr1717
Or use Stickbreaker

IraSch, 2018-08-16:
**A BIG THANK YOU! to Stickbreaker** (if there was a larger font I'd be using it).  

The combination of replacing the 3 HAL files in the Core folder, and the 2 Wire library files, PLUS using the replacement Adafruit_Si7021 library has been working for 2 days now (watch, now that I've said it, it will probably crash ... )  But seriously, getting this to work was a huge help. 

As a side note to anyone who comes across this and needs the fix: The modified Si7021 library needs the updated I2C support to work. It didn't work with the original I2C support. For this reason, I prefixed Stickbreaker's library file names both internally and externally so that the original Adafruit library wasn't effected when using other boards in the Arduino IDE. All I did was rename the files and do a "find/replace" in the 2 files themselves changing from Adafruit_Si7021 to Stkbrkr_Adafruit_Si7021 prior to installing the library.  

Thanks again.

closed 2018-10-06 by stickbreaker