stickbreaker/arduino-esp32 #49: Question about using RTOS and I2C

arminsalcin, 2019-05-29:
Hello , I have just a question not an issue .

So i started to work on one esp32 project i must use RTOS as i need to run more tasks independently , so i actually read some sensors and send to server . 

My question is as i have about 4-5 I2C devices, and suspect that in RTOS I2C get unstable and crash sometime causing module to restart. So what i do is every communication with devices over I2C i put like semaphores so only one communication at a time is available , i think that it crash  when i have more tasks reading sensors maybe two tasks reading different sensor go immediately one after another.

What is solution or is it okay to do like this , or to remove so I2C can be used in RTOS as in normal running code.

stickbreaker, 2019-05-29:
@arminsalcin The current `Wire()` implementation only supports Single Thread access.  So, you will need to create a MUTEX around all `Wire()` transactions. Each of your threads will need to grab and release the MUTEX semaphore around their transactions.

Chuck.

arminsalcin, 2019-05-29:
Thank you , you saved my project ! 

Have a nice day !

closed 2019-05-29 by arminsalcin