stickbreaker/arduino-esp32 #50: Question?

OmarAlkassab, 2019-12-25:
How to implement this i2c enhancement in my code? I'm using I2C LCD, DS1307 RTC, and PCA9555 IO Expander.

Many thanks in advance.

stickbreaker, 2019-12-25:
This repo is not up to date. All of its fixes have been merged into the main Arduino-Esp-32 repo. Use the current release instead of this one.

Chuck.

OmarAlkassab, 2019-12-25:
I tried the main repo, but the i2c bus hangs when using this 3 devices simultaneously, and didn't work until I restart the ESP32. Can you help me please?

stickbreaker, 2019-12-25:
Show me your code. 

I have I2c working with 6 24LC512 EEproms, 2 MPC23008 IO Expanders, 1 MMA8452 Accelerometer, 1 DS1307 RTCC,  Quantity is not a problem.  I2C can only talk to one sensor at a time, You have to single thread all access to I2C, and I2C cannot be accessed from an interrupt context.

Chuck.

OmarAlkassab, 2019-12-25:
Oh great! Actually I’m going to rest, I will show you my code tomorrow morning.
Can you show me the structure of your code for reading all I2C devices please? Also Can you mention me which libraries have you used (for DS1307 RTC) and if you have a suggestion for a good LCD I2C library and PCA9555 library also.
Many thanks 😊

OmarAlkassab, 2019-12-26:
[ESP Code.zip](https://github.com/stickbreaker/arduino-esp32/files/4001699/ESP.Code.zip)

@stickbreaker Dear Chuck, please find attached my reduced code (I removed unnecessary coding for simplicity)
Appreciate your help.

stickbreaker, 2019-12-26:
@OmarAlkassab I'll review you code in the next couple of days.

OmarAlkassab, 2019-12-26:
Dear Chuck, many thanks.
If you can tell me what are the tips and tricks to use multitasking of the ESP32 and the best way to use the I2C, it will be appreciated

OmarAlkassab, 2019-12-27:
Dear Chuck, please note that I use 10k pull-up resistors on the SDA and SCL pulled up to 3.3V

OmarAlkassab, 2019-12-27:
Dear Chuck,

The connection of my devices are as the following:
-PCA9555 is powered with VCC=3.3V, port0 is connected to 8 push buttons pulled up to 3.3V with 10k resistors array, port1 is connected to the gates of 8 MOSFETs pulled down with 10k resistors each one. Each MOSFET drain is connected to the second terminal of a relay, and each source to ground. The first terminal for 4 relays is connected to 5V and the contacts to 220V (Grow Light COB LED), and the other 4 relays are 12V relays (not 5V) and the terminals to 220V pumps.

-LCD I2C 20x4 is powered with VCC=5V, and its SDA-SCL pins are connected to the same bus of ESP32 and PCA9555 (also with the DS1307 RTC).

-DS1307 RTC is powered with VCC=5V, and for the Battery pin I connect CR2032 3V lithium battery.

-The board input voltage is 12V (is provided by AC to DC adapter).

After a long day with deep tests I noticed the following:
1- Without any load, everything is good for more than week (without Electricity interruption- I used backup UPS).

2- When I connect the 220VAC terminals to the relays, and after several switching on and off: sometimes the RTC stop working (the time freeze on one value) and the PCA9555 and the LCD are still working correctly. Sometimes the PCA9555 stop working (the code can’t respond to the buttons, and I cannot control the relays) and the LCD and the RTC are working correctly. Sometimes the LCD screen shows gibberish characters (sometimes doesn’t show anything) and the others are working good.

Please note that in all cases I can control other outputs (over MQTT using PubSubClient library) and monitor sensors (DS18B20, DHT11, analog PH sensor, analog TDS sensor, etc.)
[image]
[image]

I have a question here: what is cause the problem? The back EMF of the relays (even though I solder a flyback diode in parallel with each relay) or the back EMF of the 220V pumps?

I can send the connection diagram if there’s anything not clear.

Appreciate your help.

OmarAlkassab, 2020-01-01:
For whom faced this problem. I made several tests and the conclusions is that the problem is due to the switching of the relays that control an inductive load (Water/Air pumps) (Back EMF, or maybe the EMI I will check later). The ESP32 firmware is very reliable and robust.
Dear @stickbreaker I would like to hear your feedback of my code, and if there is any enhancement can be made. Closing as it is not an issue.

closed 2020-01-01 by OmarAlkassab